wavedrom/wavedrom #109: Sharing one input for multiple gates in schematic

lxkarthi, 2015-03-05:
The current JSON format of expressing a netlist restricts it to represent only tree structure. 
Generic graph structures cannot be supported. The format for representing the netlist needs to revised. Otherwise additional support for other formats needs to be implemented.

lxkarthi, 2015-03-05:
For example, representing a half adder by sharing the input ports is not possible.

drom, 2015-03-13:
Agree current implementation was done for tree structures only.
Layout engine for more complex topology will be significantly more complex.
I have implemented algorithm to minimize intersections but task need to be broken into smaller parts in order to be solvable.

lxkarthi, 2015-03-13:
Can I help implement some part of it?

drom, 2015-03-13:
Yes, for sure. I am widely open for any help in this field, and will be glad to hear your ideas, or pull your code. We probably need to perform survey of orthogonal drawing / layout algorithms?