wavedrom/wavedrom #110: If there is a syntax error (mistype), the editor crashes and cannot be reopened

dloukian, 2015-04-07:
While typing the text, I made a syntax error; the editor tries to interpret it and crashes. Not only the work is lost and cannot be retrieved, the editor cannot be started even if I delete its files in /Temp directory. This is the text with errors:

{signal: [
  {name: 'bclk', wave:    'N...|.....|...|...||......'},
{name: 'ws/frame', wave:  '01..|..0..|...|...||....1.'},
  {name: 'sdo_i2s', wave: 'x333|33333|334|443.|444333', data: ['0', 'R21', 'R20','R0','0','0','L21','L20','L0','0','R21','R0','0','0','L2','L1','L0','0','0']},
  {name: 'slots', wave:   'x2..|..2..|.2.|..2||..|.2.', data:['0','1','2','7']},
  {name: 'sdo_rj', wave: '3333333|33444|444433', data: ['L0', 'R21', 'R20', 'R19','R18','R17','R16','R1','R0','L21','L20','L19','L3','L2','L1','L0']},
],
  edge: ['']}

And this is the editor message, it does not allow to correct the error:

Uncaught node.js Error 

TypeError: Cannot read property 'x' of undefined
    at e (file:///C:/Temp/nw7720_10347/WaveDrom.js:2:11848)
    at Object.WaveDrom.RenderArcs (file:///C:/Temp/nw7720_10347/WaveDrom.js:2:12753)
    at Object.WaveDrom.RenderWaveForm (file:///C:/Temp/nw7720_10347/WaveDrom.js:2:24213)
    at Object.WaveDrom.EditorRefresh (file:///C:/Temp/nw7720_10347/WaveDrom.js:2:25767)
    at editorState (file:///C:/Temp/nw7720_10347/editor.js:61:14)
    at editorInit (file:///C:/Temp/nw7720_10347/editor.js:86:5)
    at onload (file:///C:/Temp/nw7720_10347/editor.html:19:144)

It is also unintuitive where the editor stores data and also what does it do with google analytics...

drom, 2015-04-08:
Thank you for the feedback. I will take a look.

drom, 2015-06-01:
Should be fixed in v0.11.2

closed 2015-06-01 by drom