wavedrom/wavedrom #111: Add impulses

trichner, 2015-04-09:
For timing diagrams which show fired events it is common to use something like this:

```
         ^     ^           ^
ISR1  ___|_____|___________|______
          __    __    __    __    
CLK   |__|  |__|  |__|  |__|  |__|
```
I think this would be a nice addition and I wasn't able to find it in the WaveJSON specs. Maybe use 'i' for 'Impulse'?

drom, 2015-05-28:
Great idea.

trichner, 2015-05-28:
I tried to add it myself but quickly failed xD

drom, 2015-05-28:
Do you think of only one type of impulses or multiple (green, red, black, ... up, down,..etc.)?
I know such events used in SVA: http://www.cse.scu.edu/~mwang2/verification/Sva.pdf

trichner, 2015-05-28:
up would be cool, up/down and colors would be awesome!

drom, 2015-05-28:
How about this initial set?
 * e - (EvEnt)  vertical bold black spike (no arrow)
 * s = (paSS) arrow-up green
 * f = (Fail) arrow-down red

This seems to fit SVA assertion/coverage infographic needs

trichner, 2015-05-28:
sounds good to me, maybe use the capital letters for arrow-down?

drom, 2015-06-01:
Shift button has sign ⇑ and makes capital letters. Does it make sense to associate capital letters with arrow-up? Then lower case is about arrow-down? How about non-error? what letter transformation will you associate with it?

trichner, 2015-06-03:
Yes, I also figured out this workaround a few weeks ago, but its kinda unclean solution as the arrows are not at the top/bottom and the arrow ups lead to a line on the top ;)

I guess if one would use capital letter then it should be for arrow ups.
My suggestion:

```
A/a   Arrow (black), capital for up, lower case for down pointing
S/s   Success, arrow green up/down
F/f   Fail, arrow red up/down
```

Colors would be nice, but I'm not sure if I'd be willing to spend one letter per color. Maybe some generic solution to color all kinds of waveforms? -> another feature

Alternatively we could orient us at the Tikz-timing syntax (although it is kinda silly):
http://ctan.mirrorcatalogs.com/graphics/pgf/contrib/tikz-timing/tikz-timing.pdf
with `A` as arrow up and `W` as arrow down (page 31)

ooglyhLL, 2023-05-19:
For events and assertions of any kind, I think the following states need to be represented:
1. inactive
2. active
3. vacuous pass
4. pass (and event)
5. fail

The first three can be drawn using existing levels and transitions. 4 and 5 could be represented by a vertical bar, which, at its top end would get a centered black up-pointing triangle (pass, similar U+25B4) or a cross (fail, similar U+2A2F).
I would avoid requiring colours for representing general state information and leave that for individual highlighting of information specific to the case at hand.