wavedrom/wavedrom #112: Issues with Large Waveforms

abhishek-kakkar, 2015-05-26:
I currently use WaveDrom for the web interface to my project [BeagleLogic](https://github.com/abhishek-kakkar/BeagleLogic/wiki).

[Link to the web interface (running in demo mode) here](https://beaglelogic.github.io/webapp/)

With less number of samples, WaveDrom is able to render them quickly but when the waveform grows to 10000, and the number of channels increase, the time taken to render these waveforms using WaveDrom shoots up drastically. With large datasets, it makes my browser hang for a minute.

The web interface displays the render time.

Steps to reproduce (no hardware required, as the web interface can run in demo mode as well):
 * Make the waveform larger than 10000 samples in [this file, index.html](https://github.com/abhishek-kakkar/BeagleLogic/blob/master/webapp/index.html#L418) by repeatedly copy-pasting.
 * Open the webpage and click on "Begin Capture"

I presume this is due to the increase in the size of the DOM tree.

drom, 2015-05-28:
Yes, DOM is not the fastest thing on the planet. But I will try to profile and improve the speed.
I may do partial rendering and scroll bar for the long traces. Will it help?

plysaght, 2016-06-23:
That was over a year ago.  Have we tried it yet?  It may be better now?