wavedrom/wavedrom #115: Extend the logic symbols

leoheck, 2015-07-04:
It is possible to extend the symbols including:
- A generic symbol (could be a rectangle) with N inputs and N outputs? With customizable internal label? In this way, the user can create its own symbols. 
- A way to user add new symbols.

leoheck, 2015-07-04:
It would be also nice to give support for multiples inputs on the standard symbols like in those you are using. For example with a NAND with 2, 3, 4, 5 inputs.. 

And the possibility to create complex gates like AOI (and-or-inv, without wires between the gates)

[image]

drom, 2015-07-09:
All good points.
I like the idea with complex gates. Do you have and idea how they may be described?

Now the following descripion:
```js
{ assign:[
  ["out",
    ["~|",
      ["&", "a", "b"],
      ["&", "c", "d"]
    ]
  ]
]}
```
Will generate:
[image]

Do you think that this should produce complex gate as is, or different description can be given?

leoheck, 2015-07-09:
Awesome!

I don't know, we could add some keyword indicating a complex gate. But I think you can propose something better since you know the project better than me.

```
{ assign:[
  ["out",
    ["~|c",   <------- look the c here
      ["&", "a", "b"],
      ["&", "c", "d"]
    ]
  ]
]}
```

[image]

leoheck, 2015-07-09:
It will be nice to have symbols with multiple outputs.. for example a flip-flop with Q and ~Q output.

drom, 2015-07-09:
How would you describe multi-output gate?

leoheck, 2015-07-09:
I can't find a good way by now but I am adding some symbols here to consider on the thinking process.

[image]

- A toggle and a mutex generic box components with a user-defined labels and 2 outputs.
- A Muller C-element (could also be a generic circle symbol) with 3 inputs.

I will add some ideas here when I had an idea.

drom, 2015-07-10:
We should probably separate the notion of element shape, number of pins and label.
That will require clear notation and present the challenge. Any thoughts are welcome.

YPetremann, 2017-03-07:
I think a definition system could be great, we define a gate model by it's way to be rendered, it's shape, label and ports (it could have multiple input and output port), then we instantiate the model how many time we want, finally we tell to connect edge to ports. edges could have style and label too. A port could accept many edges

'''
{
  models:{
    toggle:{
      label:"toggle",
      shape:"rect",
      ports:[
        {mode:"input",side:"left",default:true},
        {mode:"output",side:"right",name:"q"},
        {mode:"output",side:"top",name:"nq", style:"inverse"}
      ]
    }
  },
  blocks:{
    ["toggle","toggle"]
  },
  assign:[
    ["a","toggle"]
    ["toggle:q","b"]
    ["toggle:nq","c"]
  ]}
'''