wavedrom/wavedrom #116: simplify usage with RequireBin.com

drom, 2015-08-17:
WaveDrom can be used with RequireBin.com in the following way:

```js
var wavedrom = require('wavedrom')

// some parameters
var lane = {
    xs     : 10,    // tmpgraphlane0.width
    ys     : 10,    // tmpgraphlane0.height
    xg     : 120,   // tmpgraphlane0.x
    // yg     : 0,     // head gap
    yh0    : 0,     // head gap title
    yh1    : 0,     // head gap
    yf0    : 0,     // foot gap
    yf1    : 0,     // foot gap
    y0     : 5,     // tmpgraphlane0.y
    yo     : 30,    // tmpgraphlane1.y - y0;
    tgo    : -10,   // tmptextlane0.x - xg;
    ym     : 150,    // tmptextlane0.y - y0
    xlabel : 6,     // tmptextlabel.x - xg;
    xmax   : 1,
    scale  : 1,
    head   : {},
    foot   : {}
};

// add div into body
var div = document.createElement('div')
div.id = 'a0'
document.body.appendChild(div);

// added waveform
wavedrom.renderWaveForm(
  0,
  { signal:
   	[
  		{ name: "clk",     wave: "p....." },
  		{ name: "Data",    wave: "x345x.",  data: ["head", "body", "tail"] },
  		{ name: "Request", wave: "01..0." }
  	],
  	config: { hscale: 1 }
	},
  'a',
  lane
);
```
We should probably make it shorter.

drjson, 2016-01-30:
From working with the available functions in the API, I think it would be helpful for users of the library to have access to the renderWaveForm and to separate some of its parameters from the process_all call.

As a first pass, I would recommend removing the index and div id parameter and replacing it with a single reference to an element. Also the lane object would be nice if it could be removed. Additionally, the evaluator function (eva()) would be nice to have access to or to be automatically included as part of the renderWaveForm so that you receive debug information for syntax errors in the waveJSON syntax.

This would allow rendering of waveforms in a more incremental and/or interactive manner. For example you could set a text box for inputting waveJSON data and underneath have the WaveDrom rendered output visible. It would also be friendlier in general for people using jQuery or other selectors.

Modifying these functions would impact process_all as well.

drom, 2016-01-30:
Yes, I agree. The should be much simpler to use.

drom, 2016-01-30:
Well, I have updated WaveDrom to v1.2.1
now `wavedrom.renderWaveForm()` ignores the last argument and don't need the `lane` data structure.

drom, 2016-01-30:
One more update: v1.2.2

Browser script and CommonJS module exposes `eva()` API.
`WaveDrom.eva()` or `wavedrom.eva()`

drjson, 2016-01-30:
Heads up, Looks like the website is expecting WaveDrom.EditorRefresh, but the valid method is WaveDrom.editorRefresh

http://wavedrom.com/editor.html

drjson, 2016-01-30:
> When having multiple diagrams on the same page the id === 'a0' SVG has special property. It holds stash of building blocks (wave bricks), CSS attributes, and god knows what else... So, when the second diagram comes to the page it inherits all this treasure, and thus weight less.

Can you explain how this works? I looked through the code and couldn't really see what you were referring to. Are you copying the elements or inheriting somehow? Would it be possible to have those stashed in the javascript object as elements outside the viewable DOM and then you can reference them as needed. This may require a change to a multiple step rendering where you need to init or pre-render the building blocks, and then perform rendering. This could be hidden in ProcessAll for some, but would allow more flexible usage of the RenderWaveForm function.

Since you are looking at the API, I would recommend changing the WaveDrom.eva() to take either id or text so that you can use the function without an element to read.

drom, 2016-01-30:
My mistake with API renaming. I have fixed it in v1.2.3
WaveDrom.com should be fine.

drom, 2016-01-31:
Function `insertSVGTemplate`  loads one of the skins in line:17-21 with all building blocks (bricks) (SVG defines) visible globally in the DOM as the scaffolding for the picture. Then in line:28 if the `index !== 0` replace this big scaffolding with simplified version of SVG container to avoid duplicates.

drom, 2020-04-23:
LGTM

closed 2020-04-23 by drom