wavedrom/wavedrom #117: Add drag and drop reorder support

mmulet, 2015-09-07:
Hey, 
When using timing diagrams I find it really useful to be able to reorder the lane to make a close comparison between two lanes. 
I implemented it using [interact.js](http://interactjs.io/) and [snap.svg](http://snapsvg.io/)

demo available [here](http://mmulet.github.io/wavedrom/)

Let me know what you think.
Pull request is imminent.

drom, 2015-09-08:
Hi @mmulet interesting idea. And much appreciated PR. I see that size of minified package grow from 28K --> 167K. This is alarming. Do you think this functionality can be done without sansvg and interractjs?

drom, 2015-09-08:
@mmulet here some issues:
  * dependency arrows not realigned
  * signals in groups got wrong group
  * lanes go swapped not re-inserted (as one might prefer)

mmulet, 2015-09-09:
@drom: It can definitely be more lightweight.  I could use native drag events instead of interact.js, and for animation I could use something like [animateplus]( https://github.com/bendc/animateplus) which is only 10K minified. 
•	dependency arrows not realigned  - will probably just redraw the arrows
•	signals in groups got wrong group - either make it so that you can't drop outside the group, or redraw the groups after a drop( but now the groups won't all be together)
•	lanes go swapped not re-inserted (as one might prefer)- I'm not sure what you mean by this.

mmulet, 2015-09-09:
- [x] " I could use native drag events instead of interact.js, and for animation I could use something like animateplus which is only 10K minified."  The code is now 40K minified.

- [ ] dependency arrows not realigned 
- [ ] signals in groups got wrong group 
- [ ] lanes go swapped not re-inserted (as one might prefer)

drom, 2015-09-09:
* +10K is much better :+1: 
  * on lanes go swapped not re-inserted. If you take lane 2 and drop it on top of lane 5.
      * now they got swaped
      * I would expect, something like this: 3->2, 4->3, 5->4, 2->5

mmulet, 2015-09-09:
Oh, that makes a lot of sense. That seems much more intuitive, I'll definitely change to that. 
Probably by dividing the hitbox of the wavlane into a top and bottom portion, and inserting into the closest space that it is dropped.
About the empty lanes,  it would make sense to be able to insert something into an empty lane. But sometimes it's really nice to have that divider between groups. Perhaps we create a hitbox for the empty and divide it into 3 thirds, top, middle, bottom. 

a| wavelane hitboxes
------------- | -------------
 | wavelane1-top
 | wavelane1-bottom
  | empty-top
 | empty-middle
 | empty-bottom
 | wavelane2-top
 | wavelane2-bottom

if the user want to insert below wave lane1 they would drop here:

b | wavelane hitboxes                                                    
------------- | -------------
 | wavelane1-top
user drops here | wavelane1-bottom
or here  | empty-top
 | empty-middle
 | empty-bottom
 | wavelane2-top
 | wavelane2-bottom

which leads to 

c | wavelane hitboxes                                                       
------------- | -------------
 | wavelane1-top
 | wavelane1-bottom
 | new wavelane-top
 | new wavelane-bottom
  | empty-top
 | empty-middle
 | empty-bottom
 | wavelane2-top
 | wavelane2-bottom

similarly, to insert above wavelane 2 you would:

d | wavelane hitboxes                                                    
------------- | -------------
 | wavelane1-top
 | wavelane1-bottom
  | empty-top
 | empty-middle
user drops here | empty-bottom
or here | wavelane2-top
 | wavelane2-bottom

which leads to 

e | wavelane hitboxes                                                        
------------- | -------------
 | wavelane1-top
 | wavelane1-bottom
  | empty-top
 | empty-middle
 | empty-bottom
 | new wavelane-top
 | new wavelane-bottom
 | wavelane2-top
 | wavelane2-bottom

and if you wanted to just place the wavelane in the empty spot you would drop it in the middle:

f | wavelane hitboxes                                                   
------------- | -------------
 | wavelane1-top
 | wavelane1-bottom
  | empty-top
user drops here | empty-middle
 | empty-bottom
 | wavelane2-top
 | wavelane2-bottom

which leads to 

g | wavelane hitboxes                                                   
------------- | -------------
 | wavelane1-top
 | wavelane1-bottom
 | new wavelane-top
 | new wavelane-bottom
 | wavelane2-top
 | wavelane2-bottom

And there would have to be an indicator on the graph that displays where the lane will be dropped.

Last but not least, if we add the ability to fill in an empty lane, there should probably be a way to add in new empty lanes.

drom, 2015-09-09:
@mmulet these are all cool features, I like them; but before too much work is done I like to put this functionality in perspective of two main use cases:

### editor

WaveDrom editor meant for the people that like text editing more then drawing. Moving the WaveLane in the editor is just moving a line of code. Darg-n-Drop mechanism may confuse people is these manipulations will not affect the source code -- WaveJSON. Unfortunately, back-annotation of the source code may not be 100% reliable task. Taking into considerations the fact that WaveJSON is an Executable JavaScript, and user can got creative with it. I have advanced cases when user program waveform behavior in form of WaveJSON, makes it parametrizable, or use it for validation / assertion flow.

### viewer

Drag-n-Drop make more sense in the read-only mode. For example, zoom, scroll and select (highlight) are long desired and requested features. Agree, that lane rearranging is nice feature, if no semantic is broken. like we discussed above: signal grouping, dependency arrows, empty lane insertion, deletion. The question here is: should these manipulations affect WaveJSON, when user want the save it? Will all these changes be lost when user refresh the page? Will it become an partial editor flow?