wavedrom/wavedrom #121: Multiple wave per signal

Peak25500, 2015-10-06:
Hi drom,
Thanks for that tool which I was looking for for years :-)
I'm not aware of all the aspects of the software, but why not having the possibility to have multiple "wave" definition for one signal? Something like:
{ signal: [
{name 'CS', {wave: '1.0....1.', phase: 0.5, ...}, {wave: ....}}
]}
This would be nice to have different phases for the same signal at different times.
Or maybe change the phase settings?
Thanks in advance,
Peak

drom, 2015-10-06:
Thank you for the kind words.
It is technically possible with a bit of syntax cleanup. I see discontinuity problem, when waves may not stick together as one piece. In the following example: A and B combined present discontinuity in the picture:

```js
{signal: [
  {name: 'A',  wave: '0x3'},
  {name: 'B',  wave: '31x'},
  {name: 'AB', wave: '0x331x'},
]}
```
[image]

But in general it is an interesting idea and can be done. PRs welcome.

Peak25500, 2015-10-07:
Maybe option 2 seems a good idea with the "concat" keyword :-)

Why not simply use the dot "." operator on the second wave to say it should take the previous value (from the preceding wave) thus without discontinuity? But then I think it has impact on the "wave" object definition itself which should handle differently starting with a dot if concatenated or not.

haimag, 2019-09-08:
> 
> 
> Thank you for the kind words.
> It is technically possible with a bit of syntax cleanup.
> #### Option 1
> 
> Support additional options with numerical suffixes.
> 
> ```js
> {signal: [
>   {name: 'AB', wave: '0x3', phase: 0.5, wave1: '31x', phase1: 2},
> ]}
> ```
> 
> #### Option 2
> 
> Support new property `concat` ? `g` = group? ... that will do the trick.
> 
> ```js
> {signal: [
>   {name: 'AB', g: [{wave: '0x3', phase: 0.5}, {wave: '31x', phase: 2}]},
> ]}
> ```
> 
> #### Behavior
> 
> I see discontinuity problem, when waves may not stick together as one piece. In the following example: A and B combined present discontinuity in the picture:
> 
> ```js
> {signal: [
>   {name: 'A',  wave: '0x3'},
>   {name: 'B',  wave: '31x'},
>   {name: 'AB', wave: '0x331x'},
> ]}
> ```
> 
> [image]
> 
> But in general it is an interesting idea and can be done. PRs welcome.

This `Multiple wave per signal` is very useful for the differential signals and clocks (DDR) .
The multiple  waves' boundary can be used different colors or different styles(solid/dotted/dashed line)

Ganesh-AT, 2019-10-11:
The original issue and the recent September comment (DDR waveform with one regular clk_p and another dotted clk_n *in the same lane, simultaneously*) are actually orthogonal requirements. But, they can be addressed with a common proposal as described below. Can @drom give some comments on feasibility and overall impacts to legacy compatibility?

(1) Add the ability to override the skin on a per-signal basis (add a string member 'override_skin' to each signal array member, default value is the skin provided in the config object) - This can address the September comment about dotted lines - we could create a new skin that has dotted lines for selected bricks and use that for a particular lane alone.

(2) Add a wave character 'e' or 'g' - empty or gap - behavior is that nothing is rendered for the period in that lane, so the lane is transparent after rendering that span.

(3) Add ability to overlay the rendering of two signal lanes - and WaveDrom can decide that different
lanes need to be overlaid (start at the same x,y position) if the name members are literally the same. In the unlikely case that a user wants to have two distinct signal lanes with the same name, they can add invisible zero-width spaces (U+200B) at either end of the name string. This is the only part of the proposal that could break backwards compatibility.

I would love to hear your comments on this. I am slowly trying to figure out this Git development model with the PRs and all that, but I can try something out for the above set of 3 proposals. I have zero idea about SVG internal details other than that it is actually a bunch of text. Will have a bit of a learning curve figuring out (3) in the context of how WaveDrom implements the SVG rendering part, unless anyone else wants to volunteer.

Ganesh-AT, 2019-10-11:
I poked into the code base a bit more, analyzed the SVG files created by WaveDrom, and looked up the SVG specifications to understand it. I dumped out two simple clocks with two different skins:

Default:

[image]

Lowkey:

[image]

Overlaid (created by manually modifying the text in one of the above SVG files):

[image]

The approach to get this using my proposals would involve the following fundamental changes:

1. The SVG defs created by svg2json.pl need slight modification - .s$a\{$b\} needs to be .s${a}_skinname\{$b\} : make sure all skin elements have the skinname as the suffix if we want to use multiple skins in one diagram , and the g id of all should have the skinname suffix too. So, .s1 needs to be .s1_default or .s1_lowkey and so on, and g id="pclk" of type s1 needs to become g id=pclk_lowkey of type s1_lowkey / g id=pclk_default of type s1_default

2. Still trying to figure out render-wave-lane.js : appears that the xlink:href referring to the g id with cont[1][i] needs to somehow get the skin name being used - either at that point, or, as part of cont[1][i] itself. 

3. The lane.xs passed to translate in renderLaneUses needs to be fixed to refer to the xs of the first signal with that name.

This is just a rough outline of how it could be implemented - I am sure there will be pitfalls, and I welcome any comments from contributors more familiar with the code-base than I am.

drom, 2019-10-12:
I agree, that there are two different ideas floating around:
1. concatenated waveforms
2. overlapped waveforms

I would like to see more useful examples before discussing implementation details.

Maybe we should add differential signaling brick if there is a demand for it.

[image]

Ganesh-AT, 2019-10-12:
I was under the impression (based on looking at response to previous issues) that adding another type of brick is 'very costly' in terms of the number of combinations that need to be supported. The differential pair is very easily generated by the overlapped waveforms scheme. Another real-world example where overlapping would be useful is generation of signal lanes with internal delay added - See Fig. 1 and 2 of http://www.ti.com/lit/an/snla243/snla243.pdf

drom, 2019-10-13:
Here are basic examples with transition uncertanty:

[image]

[image]

@Ganesh-AT about examples from datasheets above:
It would be quite hard to support all possible types of timing diagram drawings. WaveDrom is not designed to support all of them. Any free-hand vector editor like Inkscape will be better for these cases.

haimag, 2019-10-14:
Maybe add 2 wave characters to represent differential signaling's state,
such as:
     'A' means clk_p=1,clk_n=0;
     'a' means clk_p=0,clk_n=1;

haimag, 2019-10-17:
For differential signal or clk, propose adding  4 wave characters, the true signal(pclk) function same as 'PpNn'