wavedrom/wavedrom #124: ASCII flow

drom, 2015-10-19:


poleguy, 2018-03-05:
I found wavedrom while looking for an ascii wave diagram editor for embedding in vhdl.

I think ascii output should be implemented in some basic manner. I'd say pick a simple style for implementation and do as much as you can. Even if some advanced features don't render, that's fine. In most cases you just want something simple. 

Don't let the corner cases prevent this from being implemented. Just implement maybe "10pnhlx=|." and most cases are covered.

Can I propose a five line format:

          +--------+   +----+
         /          \ /      \
    ----+  XXXXXXXX  +  DATA  +------
         \          / \      /
          +--------+   +----+

drom, 2018-03-06:
How about 3 line?
```
     /----------\ /------\
----+  XXXXXXXX  +  DATA  +------
     \----------/ \------/
```

babeihome, 2019-04-03:
It's a very important function that I would definately use it and recommand it.

proppy, 2022-11-29:
Looks like there have been some recent attempts at this:
- https://github.com/zegervdv/wavedrom-ascii
- https://github.com/Wren6991/asciiwave