wavedrom/wavedrom #126: Add the ability to add separate text fields spanning a wave break/continuation

yzmotoxer807, 2015-10-28:
When a wave break/continuation is used, the text data for that segment is combined for both sides. Add the ability to have two separate fields on either side.

My current workaround for this is to add large numbers of whitespaces in between the text to spread it over the break.

drom, 2015-10-28:
@yzmotoxer807 could you add couple of pictures for wanted and unwanted behavior?

yzmotoxer807, 2015-10-28:
The desired behavior can be generated with the following:
```
{signal: [
  {name: 'clk', wave: 'p.....|....'},
  {name: 'bus', wave: '=.=...|..=.', data: ['D0','                D1              DN-1','DN']}
]}
```

[image]

yzmotoxer807, 2015-10-28:
Related to this, I also found that there is no way to have a gap not continue the previous tick, but instead repeat the previous tick's edge, unless you are using a clock.

```
{signal: [
  {name: 'desired behavior', wave: 'p......|....'},
  {wave: '0101010|0101'},
  {wave: '1010101|1010'}
]}
```
[image]

drom, 2015-10-29:
yes, #48 is the related bug.