wavedrom/wavedrom #127: wavedrom app

huckym, 2015-11-11:
Firstly, beautiful utility! so easy to use and such beautiful results. made my task of creating timing specification a breeze. Thanks a bunch.

When I am working on an elaborate timing diagram, I'd much rather prefer to work on the local app so as to not lose my work. It would be awesome to have the capability of quickly save (ctrl-s) and load from file, even auto-save perhaps. Also being able to copy paste within the json window would be immensely useful. 

May I make a little donation for your effort, somewhere?

Thanks
Hakim

drom, 2015-11-13:
Hi Hakim,
Thank you for the kind words, and I will take a look on what can be done on the (ctrl-s) and (ctrl-c) front.
As for donations, I have added (donate) button on the main [wavedrom.com](wavedrom.com) page. You are welcome to make you contribution there.
--Alexey

huckym, 2015-11-17:
done! :)
thanks

drom, 2015-11-18:
Thank, You.
I wish more people would donate to the project then I can spend more time working on it more.
:satisfied: