wavedrom/wavedrom #129: Change font of text on arrows inside "edge" object

drom, 2015-12-12:
From discussion here:
https://groups.google.com/forum/#!topic/wavedrom/THreH-zNU7A

> I have just started using Wavedrom. Its really good.. 
> I want to increase the font size of the text inside the "edge" option. Is that possible..?
> I tried to use something like  'd->q ['tspan', {class:'h6'}, 'sometext ']',

jerryliu007, 2016-09-28:
I kinda have the need for font size request for the edge text as well. Sometimes they are just too small.

rkaletka, 2020-05-13:
bumping issue

LurenAA, 2022-12-05:
There is an acceptable solution. "Export SVG..." then change the svg file, add "font-size" to the tspan label.
`before:  <tspan>5ns</tspan>`

[image]

`after: <tspan style="font-size:13px" dy="1.4">5ns</tspan>`

[image]