wavedrom/wavedrom #132: CLI: always says "unexpected token"

smani, 2015-12-18:
I tried the CLI version using step1.js (from `test/src/`) and it always results in "Error: Unexpected token)"

    WaveDromEditor.exe source step1.js svg step1.svg

This happens on both 1.0 and 0.11.4.

smani, 2015-12-18:
The following tricks made it work:

1. Use full paths for source and output files
2. Do not keep an instance of WaveDromEditor open when executing the command

Is it possible to make the command terminate (in the command window) only after the gui closes after producing the output file?

drom, 2016-01-09:
In nw v0.12.3 I have no way knowing current working folder. Any help would be appreciated.

smani, 2016-01-11:
I have no idea about the NW. Until this is fixed, probably adding this full path requirement in the documentation would help users.

drom, 2016-01-13:
I am moving to NWJS v0.13.0
Everything is different there

drom, 2016-01-20:
I am switching to new CLI tool based on PhantomJS.
https://github.com/wavedrom/cli
Please, try it.

smani, 2016-01-24:
I followed the steps below:

 1. Install PhantomJS version 2.0 from http://phantomjs.org/download.html 
 2. Install version 4.2.4 from https://nodejs.org/en/download/ to obtain the `npm`.
 3. Run `npm i wavedrom-cli --save-dev`
 4. Run `phantomjs node_modules\.bin\wavedrom -i path\to\sample.json -p sample.png -s sample.svg`

The phantomjs command never terminates (I waited for more than 30mins).

drom, 2016-01-25:
@smani What OS are you using?

smani, 2016-01-26:
Filed a new issue: https://github.com/wavedrom/cli/issues/2

It is a Windows 7 64 bit OS.

closed 2016-01-30 by drom