wavedrom/wavedrom #133: Custom (non-transitioning) Waveform Request

fraser125, 2015-12-24:
I'm trying to create a visualization of a single wire protocol (N64 Controller) with the following basic shape:
[image]<br>
The line is pulled up by default and pulled down for the data bits as pictured. There are also 2 different Stop Bit patterns, so I need a total of 4 custom waveform shapes. <br>
In order to implement this I tried creating a custom skin, but couldn't seem to get what I wanted. For example element 000 is used twice for the 0 waveform. if there were 2 different pieces used for the low I could create the low portion as one piece and the high portion as the second. <br> 
I'm not a Javascript programmer, but it seemed like good code and I found where the changes might go but thought I should at least ask if this is the right direction, before proceeding.
Thanks.

drom, 2015-12-25:
Here is an example of how it can be done:
```js
{
  signal: [
    {name: 'a', wave: 'hlh..l..hl..h'},
    {name: 'b', wave: '101..0..10..1'}
  ],
  head: {tick: [,,,'"1"',,,,'"0"',,,,'"0"']},
  config: {skin: 'narrow'}
}
```
[image]

fraser125, 2015-12-25:
Thank you for your help, I used a combination of the script in your response with a little adjustment of the skin and I'm very happy with the result. The adjustment was the 1mx transition. 
``` javascript
{signal: [
  {name: 'Zero', wave: 'xl..hx'},
  {name: 'One', wave: 'xlh..x'},
  {name: 'Console Stop Bit', wave: 'xlh.xx'},
  {name: 'Controller Stop Bit', wave: 'xl.h.x'},
],  config:{skin:'n64'}, }
```

[image]

closed 2015-12-25 by fraser125