wavedrom/wavedrom #134: Specifying a phase will ruin the alignment of the wave

anderswb, 2016-01-14:
If I add a phase to a signal, it seems like the grphics are not aligned at the beginning and end of the wave:

```
{signal: [
  {name: 'clk',   wave: 'l.p.....l'},
  {name: ' tst',  wave: '10......1', phase: 0.15},
 ]
}
```

[image]

drom, 2016-01-15:
The reason being, that each lane constructed from the blocks of the standard size, like bricks in the wall. Phase shifts all the brick raw, but may create gaps in the beginning or the end.

rosholm, 2017-06-12:
I would highly appreciate if the tool could extend the beginning and the and so that they align.
It will make the diagrams look much prettier when using phases, and make WaveDrom an even more attractive tool.

I have no idea how hard it is to add this feature, but maybe a solution could be to:
1. Let the "phased" signals begin left of time = 0 and have the signal names be a text box (at least one clock period wide) with solid white background which are placed on top of the signal and thus hides the signal left of time = 0
2. Have a one clock period wide solid white rectangle block the (right) end of the timing diagram. The position of this white rectangle could be determined by the end of the shortest or longest "non-phased" signal.

Somehow, extending in the beginning already works when the phase is 0.5:

`{signal: [
  {name: 'clk',         wave: 'p.....'},
  {name: 'rd_ack',      wave: '0.10110', phase: 0.5},
 ]
}`
[image]

Ganesh-AT, 2023-10-28:
I believe the best way to solve this problem would be to use the 'clip-path' SVG feature. I implemented it in my private version, but am running into some browser compatibility issues - it works great for 2D bricks, but is not good for 1D ones. Example screenshot:

[image]

The first brick I want to render for the CLK in the first lane is properly clipped in Firefox (middle window), but is not rendered properly in Edge or Chrome. Unfortunately, Firefox fails to render tspan sub text properly - and in general, the WebKit rendering (Chrome / Edge) rendering is much better for creating PDFs from WaveDrom SVGs.

Essentially, the approach would be something similar to this. I am still in the 'verification' stage to ensure that existing diagrams are not messed up due to this update:

(1) In 'parseWaveLane' : The unseen bricks set should optionally ignore the last of the currently unseen bricks (based on the phase coming in)

[image]

(2) Modify renderWaveLaneUses to take in a 'clip path percentage' parameter and use it to set the clip-path polygon

[image]

(3) Pass in the new cpp_pct parameter in the call after computing it based on the original phase and already available variables that consider hbounds:

[image]

I would love to submit a PR, but the ones I sent in for sub-cycle bug fixes back in 2019 are still not merged in. In the meanwhile I added a large number of features / additional bug fixes that have made my code base (from which the above screenshots have been captured) diverge a lot from the version here. So, I will leave it to someone else with a code base closer to the current master to utilize the above approach for issue resolution.