wavedrom/wavedrom #138: EPS export

drom, 2016-02-14:
Microsoft Word can't handle SVG, but accepts EPS.
Export should be possible.
Here example of PS project.
https://github.com/jbush001/DrawWaveform

polpo, 2016-02-14:
AFAIK, word can place an EPS file, but doesn't actually interpret the PostScript. It will send the PS to a PS printer or PDF, but for any other output format it uses the preview image attached to the EPS, which is usually very low resolution.

Word does understand the WMF and EMF formats as vectors, though.

drom, 2016-02-14:
@polpo thank you for the information. I am fine with M$ forwarding EPS to the printer. Can we sneak in hi-res preview into EPS?
I am not big fan of WMF, EMF, but have you seen ant good docs on the format?

drom, 2016-02-15:
@polpo here is one more thread on the issue: https://groups.google.com/d/topic/microsoft.public.mac.office.word/nxLC9R1r4jg/discussion
Looks like you are absolutely right. EPS is not an option, because M$ may print bitmap or vector depend on the printer.
WMF/EMF is a can of worms, an I don't even want opening it. So, pass for now.

drom, 2016-03-13:
Duplicate of #37

closed 2016-03-13 by drom