wavedrom/wavedrom #140: Not displaying

audreyong, 2016-02-25:
Not sure if I'm doing this right but I'm running windows, and I downloaded wavedrom-editor-v1.2.0-win-ia64.zip. After extracting it, I ran WaveDromEditor.exe. I see the editor and change the text, but no diagram appears on the upper half of the screen. I'm able to export the png fine though, even though I can't see what diagram I'm making as I type it.

drom, 2016-03-03:
What version of Windows do you use?

audreyong, 2016-03-03:
Windows 7 Professional

drom, 2016-03-13:
I just checked on Windows 7, and I see the rendering like this:
[image]

I do see abnormal behavior sometimes after start (~20% of starts).
The scroll-bar appears on the right and rendering is not happening.
[image]

For me it can be fixed by resizing an application window, Menu -> Rotate Layout.

Is it similar to your case? Can you attach a screenshot?

audreyong, 2016-03-13:
The second screenshot is exactly what I'm getting. (I can attach a screen shot if you still want it, but I don't think it'll provide any additional information.) Once, when I opened WaveDrom, I got the first screenshot, but I would say that most of my experiences are of the second screenshot.
I tried Menu -> Rotate Layout, and it does fix the issue for each startup.

drom, 2016-03-14:
Good. I am porting editor to NW.JS v0.13. Will be back after this.

drom, 2016-07-14:
considered as fixed. reopen if any concerns.

closed 2016-07-14 by drom