wavedrom/wavedrom #142: data does not shift with phase

itsayellow, 2016-03-02:
Thanks for wavedrom, it's great!

My issue is that when I phase shift a signal that has a data field, the data does not shift with the waveform.

For example, the attached testcase should make the same waveform but with different phase shifts.  Instead, while the color is shifted with the phase, the numbers are not shifting with the phase, which makes the colored data change value.

I think it make more sense to have the data shift along with the waveform.  In my case I want to replicate the same circuit with a different phase, and it would be nice to reuse the exact same waveform syntax--only changing the phase directive, and have everything shift along with the phase.

[testcase.js.txt](https://github.com/drom/wavedrom/files/155397/testcase.js.txt)

drom, 2016-03-03:
Yes, good catch.

itsayellow, 2016-04-27:
I have a fix for this that I think is working.  What's the best way to submit the fixes?  Should I do a pull request?

drom, 2016-04-28:
@itsayellow yes, please send PR.

closed 2016-05-13 by drom