wavedrom/wavedrom #144: Wavedrom in mediawiki

Martoni, 2016-03-09:
Does somebody know how to integrate wavedrom in mediawiki ?

drom, 2016-03-10:
@Martoni It would be cool :+1:
It should not be too difficult to do. Do you have good mediawiki plugin template? example?
Does anybody have access to mediawiki installation to try?

Martoni, 2016-03-10:
@drom Yes I have an access to a mediawiki installation. I've got a private mediawiki. But I don't know how to make a plugin template.
I'll keep trying to do it, if I find how to I will explain it here.

Martoni, 2016-03-11:
@drom I wrote a [quick&dirty plugin](https://github.com/Martoni/mediawiki_wavedrom). It works for me, but not perfect.

closed 2016-03-11 by Martoni

drom, 2016-03-11:
Great job. That was quick!

Martoni, 2016-03-11:
and dirty ;)

drom, 2016-03-11:
It is fine. Keep working and you will make it clean :smirk: