wavedrom/wavedrom #146: Clock Skew / Asynchronous Clock Support

acarlson1029, 2016-03-17:
Hi,

I was wondering if it would be possible to support some kind of skew or multiple-clock diagramming on the same waveform. This is especially useful for diagrams dealing with clock-domain-crossing, or showing off C2Q delays through elements.

Ideally you could specify ratios between the "main" clock and any other clocks.
Or you could register a clock as a separate type of signal that any other signals can reference.

Rough example:

```javascript
{ signal: [
    {clk: 'clkA', wave: 'p......', ratio:1},  // default is 1.0
    {clk: 'clkB', wave: 'p..........', ratio:0.66},  // clock edge distance is 66% of the default clock
    {clk: 'clkC', wave: 'p......', skew: ['clkA', 0.05]},  // +5% skew from clkA
    {clk: 'clkD', wave: 'p......', skew: ['clkA', -0.10]}, // -10% skew from clkA
    {name: 'datB', wave: 'x223223', clk: 'clkB'}, // align to edges of clkB
]}
```

acarlson1029, 2016-03-17:
Nevermind!!

I didn't read the tutorial closely enough, the skew is already supported (`phase`) and the `period` argument can handle spacing out clocks and assigning child signals to specific clock edges.

Sorry about this. Great stuff.

drom, 2016-03-18:
Great to hear. Closing.

acarlson1029, 2016-03-18:
I started digging into this a little more closely. It looks like the `period` argument is limited to an integer value?

I'd love to be able to specify floating point clock ratios here.

drom, 2016-03-18:
@acarlson1029 you are right. Image is constructed from blocks of standard size (bricks), so the `period` rounded to the brick size.
Depend on the scenario certain period ration can be more or less illustrative.
`{period:1}, {period:3}` -- fast & slow clock
`{period:2} {period:3}` -- similar frequency clocks.
`{period:2} {period:3, phase:0.25}` -- similar frequency but unrelated clocks. Never in sync.
you can use `config: {skin:'narrow'}` to compress diagram a bit.

itsayellow, 2016-04-10:
@acarlson1029 I think wavedrom is rounding to 0.5 tick resolution for edge placement, so a bit better than integer.  But yes, you can specify floating point, but every edge will end up on the nearest 0.5 tick.

closed 2019-07-14 by drom