wavedrom/wavedrom #147: Mac OS X Application named "nwjs", no way to change to WaveDromEditor

itsayellow, 2016-03-24:
This is a minor and cosmetic issue, but a bit odd.

The name of the WaveDromEditor application icon in Mac OS X always shows up as "nwjs" in Finder or Launchpad. 

I tried to rename it to WaveDromEditor, but it still remains "nwjs".  In the Terminal, it shows up as WaveDromEditor.app. (!)

It must be something inside the .app bundle that makes any GUI interpret the name as something other than "WaveDromEditor" .

itsayellow, 2016-03-28:
Figured out how to fix it.  
In file:
WaveDromEditor.app/Contents/Resources/en.lproj/InfoPlist.strings
you just need to change the line from
CFBundleDisplayName = "nwjs";
to:
CFBundleDisplayName = "WaveDromEditor";

itsayellow, 2016-03-28:
https://kb.applingua.com/2011/10/how-to-localize-app-names/

drom, 2016-03-28:
Thanks. I have no OSX to try, but I will try to incorporate this change into the next build.

drom, 2016-04-04:
@itsayellow could you try v1.3.1 ?

itsayellow, 2016-04-04:
Looks great.  Installs as "WaveDrom" and shows up both in Finder and in Launchpad under that name.

closed 2016-04-04 by itsayellow