wavedrom/wavedrom #148: sub-cycle pulse insertion capability

drom, 2016-03-28:
Hi, Gents -
 
WaveDrom is a cool tool for creating cycle-based waveforms, but lacks the ability to place sub-cycle pulses into the waves (without lots of extra characters to stretch out the cycle). I would like to see sub-cycle pulse insertion capability and I would also like to propose the following syntax that could be used (I don't care what the syntax is, but I thought I would at least make a proposal).
 
For scalar signals, could I suggest:
 
{name: 'din', wave: '01.0<010.>..10' }
This would insert a ¼ cycle pulse into the 5th cycle. The syntax <010> would also be equivalent to <010.>
The inclusion of the <> angle brackets could denote ¼ sub-cycle granularity.
 
For buses, a similar syntax would be:
 
{name: 'data', wave: '22.<.x2.><.22.>222', data: [F0 E3 E7 12 24 A1 B1 B2]}
This would insert X|E7 into the 4th cycle and 12 |24 into the next cycle. Again, the syntax <.x2> = <.x2.> and <.22.> = <.22>
 
You might also allow <<>> to denote 1/10th cycle pulses, but the <> syntax for ¼ cycle pulses is a great start and might be enough.
 
The ability to insert short pulses into a cycle-based waveform is a common need, something that would have helped me when using WaveDrom.
 
Cool tool. I hope you will consider this enhancement.
 
Regards - Cliff

drom, 2016-03-28:
Hi, Cliff

Thank you for the kind words, great idea, and syntax proposal.
Agree, that it addresses a common need, and it was requested several times in the past.

It will be quite easy to implement 1/2 of a cycle pattern for the default (period:1),
because each cycle of diagram actually constructed out of 2 bricks (skin references).
As well as: 1/4 for (period:2); 1/6 for (period:3); etc.
Is it fine for the first version?

I like your syntax <.10.> and the way to avoid tail padding <.10>

We will definitely consider implementing it soon. Can I mirror this request on the GitHub here: https://github.com/drom/wavedrom/issues ?
That is our usual place for bugs/enchantment requests; public for all collaborators.

Regards,
Aliaksei

drom, 2016-03-28:
Hi, Aliaksei -

The 1/4 for period:2 should work fine. 

Permission granted to mirror the request.

Regards - Cliff

drom, 2016-05-19:
@itsayellow what do you think of this idea?

drom, 2016-05-19:
Other option might be using nested arrays of data points. Instead of:
```
{
  name: 'data', wave: '22.<.x2.><.22.>222',
  data: ['F0', 'E3', 'E7', '12', '24', 'A1', 'B1', 'B2']
}
```
Use something like:
```
{
  name: 'data', wave: '22.$$222',
  data: ['F0','E3', '. x E7 .', '12', '24', 'A1', 'B1', 'B2']
}
```

itsayellow, 2016-05-19:
I like that, because it preserves the "one character equals one time period" methodology.  It would make the code to parse data a little trickier, but I think that should be ok.

One question, I'm wondering if this would fulfill everyone's needs for sub-cycle waveforms.  I could imagine wanting to put a tiny pulse in a sub-cycle waveform, and that wouldn't be in the form of a data label.  Could you put 0 or 1 in the subcycle data list?  If you did, then it would be a little strange because then data field takes on an overlapping meaning with the wave field that it didn't used-to have.

itsayellow, 2016-05-19:
Here's another way that I haven't thought too hard about whether it's great or not, but I thought I'd throw it out there.

{
  name: 'data', wave: '22.ab222,
  subcycle: { a:'01.0', b:'10.1' }
}
This allows little snippets of sub-cycle waveforms that you could re-use, but it I'm not exactly sure how you would incorporate data strings into this.

Although could you even read sub-cycle data labels?  As it is, if there's only a half-cycle of data, wavedrom doesn't even try to print the label (if it's in the start or end of a waveform, for example.)

itsayellow, 2016-05-19:
I suppose you could bring it all together like
```
{
  name: 'data', wave: '22.ab222,
  subcycle: { a:'01.0', b:'0220' }
  data: ['F0','E3', ['12', '24'], 'A1', 'B1', 'B2']
}
```

drom, 2016-05-19:
I agree that you need explicit subcycle format string to be compatible with `wave` format.
Here is another option:
```js
{
  name: 'data', wave: '22.$$222',
  data: ['F0', 'E3', ['.x2.', 'E7'], ['.22.', '12', '24'], 'A1', 'B1', 'B2']
}
```

itsayellow, 2016-05-19:
I think I like that a little better.

I was trying to think of a way you could re-use sub-cycles, allowing you to re-use a standard runt-pulse for example.  But using letters in the wave field is a little unsatisfying because we already use 'h','l','H','L','p','P' and it's a little weird to just skip over them and assume the other letters are sub-cycles.  In that way the $ is a nice clear identifier.

We might do like above and also use a "subcycle" field to define all the $ time-periods in order, like the "data" field currently defines all the data cycles:
```
{
  name: 'data', wave: '22.$$222',
  subcycle: ['.x2.','.22.'],
  data: ['F0', 'E3', ['E7'], ['12', '24'], 'A1', 'B1', 'B2']
}
```

That way data is only used for data, and subcycle looks like a list of "wavelets" defining each $ in sequence.

drom, 2016-05-19:
@itsayellow the more I look at the original proposal from Cliff Cummings, the more I like it again.
The main concern is that we are stretching `wave` sequence and (1 char !== 1 cycle) anymore.
The benefit is all format is in one place, and it is the most compact form.

itsayellow, 2016-05-20:
It's true.  It is the conceptually simplest solution.  And everything is in-line, you don't have to pop your eyes back and forth within the same wave object to find what each subcycle is in a separate definition somewhere.  That to me is the biggest advantage.

Already on my own I make little javascript code snippets to do things with my waveforms, like "x4()" which repeats the same string 4 times, etc.  So I have broken the direct 1char<->1cycle wave relationship by adding little functions and it is ok.  Not as easy to read, but ok.  It's easier if you break the 1<->1 in the same place in every waveform, then they all line up.

This would also work for this proposal--if there is a region on the x-axis where EVERY waveform needs more detail, then all the "wave" fields for all waveforms would still line up perfectly, e.g.
```
{
  name: 'wave1',
  wave: '0...1.<010101010101>..0..1'
}
{
  name: 'wave2',
  wave: '0...1.<000001.00000>..0..1'
}
```
So you could keep things tidy if you tried.

drom, 2016-05-20:
@itsayellow One limitation of this concept we have to convey clearly to users that number of subcycle details would depend on period.

period: N --> 2*N subcycles

closed 2016-07-06 by drom

drom, 2016-07-06:
Hi Cliff,
In fact I did. Please try the code below in online version here: http://wavedrom.com/editor.html
There is certain quickness in rendering of sub-cycles, let me know if it is close to your expectations.

--Aliaksei

```js
{ signal: [
  {
      name: 'P1',
      wave: '==2<30>2<0xx1>2333444555',
      data: '0 1 2 a b c 3 4 5 d e f 6 7 8'
  },
  {
      name: 'P2',
      wave: '=2<15.1>2<01>2355',
      data: '0 1 2 a b c 3 4 5 d e f 6 7 8',
      period: 2
  },
  {
      name: 'P3',
      wave: '=2<1001.1>2<01.5>3',
      data: '0 1 2 a b c 3 4 5 d e f 6 7 8',
      period: 3
  },
  {
      name: 'P4',
      wave: '=2<10zuzd1x.1>2',
      data: '0 1 2 a b c 3 4 5 d e f 6 7 8',
      period: 4
  }
]}
```

> On Tue, Jul 5, 2016 at 1:53 PM, Cliff Cummings <cliffc@sunburst-design.com> wrote:
> Hi, Aliaksei -
>
> Did you ever add this capability to Wavedrom?
>
> Thanks - Cliff

commit 4a2c22d8 (2016-07-06): fixes #148; sub-cycle pulse insertion capability