wavedrom/wavedrom #150: more colors

itsayellow, 2016-04-18:
An enhancement request for more colors.  At least one more, like a nice light green would be cool.  For "6" and up.

I do a lot of timing diagrams with signal flowing through processing, and I like to visually see the same color at each stage of processing to show the same data progressing through the pipeline.  Another color would be helpful.

This might be another "help wanted" situation where I can try and look at the code, but at least I wanted to register this enhancement request officially.

drom, 2016-05-13:
@itsayellow currently colors are part of skin. Example: https://github.com/drom/wavedrom/blob/master/unpacked/skins/default.svg

Adding more colors means expanding skin. Which is possible.

piotrva, 2019-01-06:
I also vote for the enhancement. Maybe it will be possible to make colour independent from skin?

drom, 2019-11-28:
implemented 4 more colors 6,7,8,9 https://github.com/wavedrom/wavedrom/commit/1d4d25181d6660b5d069defcf04583158b51aa5c

closed 2019-11-28 by drom