wavedrom/wavedrom #154: hbounds to specify left,right boundaries

itsayellow, 2016-05-09:
referencing: https://github.com/wavedrom/wavedrom.github.io/issues/13

I have config:{ hbounds:[hmin,hmax] } pretty much implemented on my hbounds branch on my fork of wavedrom.

I shift the phase of all waveforms to implement hmin, and then adjust xmax if hmax is smaller than xmax would be computed from waveform length.  It seems to work pretty well with the rest of wavedrom and not mess things up.

One thing I'm not sure about: head.tick, head.tock, foot.tick, and foot.tock can all be different specifications for timing numbering.  Originally I was going to base the hbounds hmin and hmax on the absolute number of each time period, but since there could be 4 different disagreeing time reference numbers, currently I implement hbounds as defined by time periods starting at 0 for first time period, ignoring what tick and tock numbers are.

I think this is a pretty good compromise, but I'd be open to other ways to specify hmin and hmax.

If the current config works, then I'm pretty much ready for a pull request to add this feature.

itsayellow, 2016-05-10:
Forgot to mention, I also make sure the ticks and tocks are correct on the new visible part of the waves if the user specifies a non-zero hmin.

itsayellow, 2016-05-13:
The new hbounds code in v1.3.0 has a couple of bugs (arcs and gaps are not in the right place.)  I think I know how to fix it but will have to wait until I get home tonight.

drom, 2016-05-13:
@itsayellow yes, arcs would be the hardest to do. We may need to remove them completely when one of two nodes are out of scope.

itsayellow, 2016-05-13:
Actually the arcs are still shown off the diagram (sometimes over the signal names) just by varying the phase of a particular signal.  (This might be considered a separate rendering bug.)

In my user-space hbounds I actually doctor the node string to remove letters that are not visible due to phase.

itsayellow, 2016-07-29:
Should we consider this bug closed?  If we're worried about arcs off-diagram showing up, I think that might be considered a separate bug, since it can occur with any phase change on a waveform.

drom, 2016-07-31:
@itsayellow yes, closing

closed 2016-07-31 by drom