wavedrom/wavedrom #157: Add register description generation function

Thoragan, 2016-05-20:
It may be very usefull to have a tool, that will generate register description picture like in [NXP documentation](http://cache.nxp.com/files/32bit/doc/ref_manual/MPC5553_MPC5554_RM.pdf?fpsp=1&WT_TYPE=Reference%20Manuals&WT_VENDOR=FREESCALE&WT_FILE_FORMAT=pdf&WT_ASSET=Documentation&fileExt=.pdf) (page 199).

[image]

It seems to be very similar with geneating timing diagrams and will be nice addition to WaveDrom.

drom, 2016-05-20:
@Thoragan the first shocking thing on this picture (to my LE brain) is BE order of bits ;)

drom, 2016-05-20:
@Thoragan in fact I have standalone package with this functionality: https://github.com/drom/bitfield
I will work to make it more accessible. Please file all feature requests there.
It is extraction from old version of WaveDrom and at some point I am going to integrate it back.

drom, 2020-04-23:
LGTM

closed 2020-04-23 by drom