wavedrom/wavedrom #158: Feature Request: Absolute Timing Markers

jwcxz, 2016-07-06:
Oftentimes, it is useful in a diagram to clearly mark certain absolute times and reference them later.

A simple mockup (zipped because GitHub isn't letting me attach a PDF):
[example-mockup.zip](https://github.com/drom/wavedrom/files/349731/example-mockup.zip)

In this example, I've created two markers, T1 and T2 at tick 2 and tick 4 respectively.  A bonus nice-to-have would be support for coloring the marker line with a custom color as shown in T2.

Maybe the syntax could be something like:

```
{
    signal: [...],
    edge: [...],
    marker: [
        { text: 'T1', tick: 2 },
        { text: [ 'tspan', { fill: 'red' }, 'T2' ], tick: 4 }
    ]
}
```

Perhaps it could support putting markers on tick or tock.

I found a similar request [here](https://groups.google.com/forum/#!searchin/wavedrom/marker/wavedrom/49XVpJ-dJKc/Gr2VPkre2IMJ) but didn't want to revive a dead thread.

drom, 2016-07-13:
@jwcxz inserted your example as Bitmap:

[image]

heartacker, 2021-08-19:
that will be great if we could specify the position of marker
```jsonc
{
    signal: [...],
    edge: [...],
    marker: [
        { text: 'T1', tick: 2 },
        { text: [ 'tspan', { fill: 'red' }, 'T2' ], tick: 4, pos: 0.1 } // default pos:0.5
    ]
}
```