wavedrom/wavedrom #160: Feature: more options on edge specifications

itsayellow, 2016-07-29:
issue https://github.com/drom/wavedrom/issues/158 got me thinking about annotating diagrams.  Here are some of my thoughts so we can start a discussion

If two nodes are very close together, the spline connecting them becomes nearly a straight line, seeming to come out of the side of the arrowhead.  It would be nice if the line connecting them could be specified to be "s-shaped" so that edge lines can start and finish horizontally in such cases.  Possibly the format for the edge could be "~~" indicating a spline that is structured this way  An example of a tight timing where the line comes out of the arrowhead's side:
[image]

I have some fairly involved timing checks in my simulation testbench, and I was thinking I want to use wavedrom as an output from my simulation, showing setup/hold violations.  Something simple to aid this would be an indication of an error, like a red edge line/arrow instead of blue.

Sometimes if two edge descriptions are close together they overlap each other, one obscuring the other.  Maybe for text we could use "<" or ">" before the word to indicate the word should be to the left or right of the edge line respectively, instead of centered all the time as now.
Here are overlapping edge id texts:
[image]

I tried to look at the edge code before but couldn't make much sense of it, but maybe with enough looking I could figure it out.  Unfortunately I am ramping up to tape-out right now with not much spare personal cycles.  :)  I at least thought I'd write down these thoughts and maybe I can start to look into it despite my eventful work life.

itsayellow, 2016-08-02:
It looks like lib/arcs.js and lib/arcs.new.js are completely unused.  Is that correct?  If so should we delete them?

I'm guessing that currently lib/render-arcs.js is where all the arc rendering stuff happens, correct?

drom, 2016-08-09:
@itsayellow yes, I have started changes to the arcs system long time ago. Haven't finished. These two files are old attempts of refactoring. New system supposed to support very complicated arcs. https://github.com/drom/wavedrom/wiki/WaveJSON

itsayellow, 2016-08-09:
Oh darn. I wish I'd known that since I just spent a lot of time working on
them. Well I guess I'll hold on to my work somewhere and wait for you to
commit your changes. (?)

drom, 2016-08-09:
@itsayellow sorry for confusion. I have started my changes so long ago that it is ash my now. We can easily go with your code. We can just enhance your code with the proposal. Do you understand the gist of new arrow proposal?

itsayellow, 2016-08-16:
I like the flexibility of this.  My only thought is that I'm not sure I would need this much flexibility.  There are some other questions like the curvy shape: how does the arc interpreter know whether to make it concave-up or concave-down if it's part of a multi-segment run?  Maybe we'd need two characters for the curve, like maybe the left, right parentheses?  So that one paren is concave-up and its mirror-image paren is concave-down?

Before I saw your proposal I was thinking more along the lines of having two characters for "first half" and "second half" of arrow.  This would make implementation a lot easier and cleaner.  We could still do all of the different types of lines, but it would maybe only have two segments.  Then we might also be able to pick the anchor point at x.start+dx/2, y.start+dy/2 between the two segments for everything except horizontal and vertical lines.

My main work over the past weeks was to make curvy arrows always be curvy, for example when start and end of arrow are directly above/below each other, the arrow ends up being "S-shaped" so that start and end of arrow are still horizontal.  I'll post a picture when I get back to my laptop that has my work.

itsayellow, 2016-08-16:
I guess one advantage of multi-character arcs is that you have useful flexibility of where to place the text label, even if the arc characters are all the same.