wavedrom/wavedrom #161: How to convert JSON to SVG from command line?

ScumCoder, 2016-07-30:
I want to integrate WaveDrom into LaTeX. AFAIU this can be achieved by invoking the `WaveDromEditor` executable, feeding the JSON to it, getting the SVG from it, and pasting this SVG into LaTeX. I'm pretty sure I'll be able to do this using the `\begin{python}...\end{python}` environment.

The only thing I need is a set of command line arguments that can be passed to the `WaveDromEditor` (or maybe some other) executable to convert JSON text into SVG file.

drom, 2016-07-31:
@ScumCoder here is CLI version of WaveDrom: https://github.com/wavedrom/cli
Will love to know how it went.

closed 2016-07-31 by drom

ScumCoder, 2016-08-01:
@drom thanks a lot and sorry for bothering you, I don't know the GitHub structure yet and couldn't find this repo on my own