wavedrom/wavedrom #165: Command line interface

Bamban, 2016-11-10:
Hi ! First of all thanks a lot for this awesome tool.
I'm using it to generate waveforms for some documentations. I currently have some json wave files in a directory. In order to process them with latex, I need pdf. I can easily go from svg to pdf (which is great because everything is vector) with inkscape. The only thing I can't automate is the export of json to svg. I currently open the editor and click "export svg".

Hence my question : would it be possible to imagine a command line interface doing something like this ?
`./wavedrom --in wave.json --export-svg --out wave.svg`

Thanks for your time !

drom, 2016-11-10:
Hi @Bamban . Thank you for the kind words. Here is CLI version of wavedrom https://github.com/wavedrom/cli Please try.

3DES, 2019-02-27:
CLI is really a great idea!

Any chance to get it implemented directly into the standalone waveformeditor, since the editor has all needed options already implemented and especially since PhantomJS project has been suspended?

Thanks a lot spending your time!

drom, 2019-02-28:
Yes, It is possible since `chromium` has `headless` mode now: https://developers.google.com/web/updates/2017/04/headless-chrome

And if `NWJS` supports it I will implement CLI for it.
https://github.com/nwjs/nw.js/issues/769