wavedrom/wavedrom #166: Please add an additional color field per wave.

Paebbels, 2016-11-19:
Each wave has additional description fields for texts, edges markers, etc.
Please add a "color" field, which applies a color to each wave section as "data" applies a text in the current version.

----------------------

I'm going to use wavedrom in "sphinx-wavedrom" and ReadTheDocs.

drom, 2016-11-19:
@Paebbels Do you mean, stroke color for the actual waveform drawing?

Paebbels, 2016-11-19:
Depends on how complex it is, I would be happy with more colors for data (`=`) parts.

One a second thought it would be better to have:
* `color` -> line color
* `fill` / `fillcolor` -> for the filling in e.g. `x` or `=`

Fill color for e.g. 1 could mean to fill the wave from baseline upto the line (some simulators do so to distinguish 0 (line and no filling) from 1 (line with filling). This is easier to read with long flat lines.

This would add a very flexible feature to wavedrom and extend the current colors (2,3,4,5).

**Examples:**

This is GTKwave showing a simple simulation. GTKwave uses no "shadows" (filling):
[image]

This is a screenshot from Xilinx ISE Simulator. It's easier to read the wave, because the 1 has a shadow:
[image]

drom, 2016-11-19:
@Paebbels 
WaveDrom has the database (skin) that defines the drawing of the building blocks (bricks) with all shape / color attributes. Here is default skin: https://github.com/drom/wavedrom/blob/master/unpacked/skins/default.svg

WaveDrom just assembles final picture from the references to these bricks. This is done to save space.

WaveDrom can't randomly modify the drawing aspects of these bricks like color.
You can modify style of drawing of diagram by customizing the skin and then translating skin SVG -> JS . You may extend number of colors but it has n^2 effect on the skin size.

Paebbels, 2016-11-19:
Hmmm,

but all SVG snippets are css-styled.

```xml
  <g
     id="0mz"
     transform="translate(200,80)">
    <path
       sodipodi:nodetypes="ccc"
       inkscape:connector-curvature="0"
       d="m0,20 3,0 C 10,10 15,10 20,10"
       style="fill:none;stroke:#000;stroke-width:1;stroke-linecap:round;stroke-linejoin:miter;stroke-miterlimit:4;stroke-opacity:1;stroke-dasharray:none"
       id="zmo_120_40-2-7-6" />
  </g>
```

As far as I remember, CSS is editable from HTML via JavaScript and DOM. So the JavaScript code can apply other e.g. stroke colors while assembling the waveform.

drom, 2016-11-19:
@Paebbels If you look into the DOM of the diagram you will see the series of references:
```html
...
<use xmlns:xlink="http://www.w3.org/1999/xlink" xlink:href="#xxx" transform="translate(0)"></use>
<use xmlns:xlink="http://www.w3.org/1999/xlink" xlink:href="#xxx" transform="translate(20)"></use>
<use xmlns:xlink="http://www.w3.org/1999/xlink" xlink:href="#xxx" transform="translate(40)"></use>
<use xmlns:xlink="http://www.w3.org/1999/xlink" xlink:href="#xxx" transform="translate(60)"></use>
...
```
So you can't style different references to the same brick differently. If you change CSS of the brick it will change all places where these brick is referenced.

Paebbels, 2016-11-19:
Oh...

hickninja, 2018-05-07:
@drom - you can specify the fill as 'inherit' for the data types, and then use 'class="whatever"' on your <use> statements to set the fill style. So for all the bricks with single fill you could simplify to having just one copy of them in the skin (instead of one per color).

For the 2-color fill blocks (the ones transitioning from one color to another, you can use the color and fill technique [here](https://tympanus.net/codrops/2015/07/16/styling-svg-use-content-css/). For that technique, you don't set a fill on the first path, and you set the fill on the second path to 'currentColor.' Then you set a class on your <use> statement that includes styling for fill and color. This would let you replace all the two color bricks with a single one.

This would eliminate like half the bricks in the skin, at the expense of adding classes to all your use statements. But you would then be able to define a ton of colors. Bonus points if you allow a "colors" array in the config object for user-customized colors...