wavedrom/wavedrom #172: How to delete the last pasted codes?

Square2017, 2017-03-27:
Please describe your system environment before submitting your Issue.
Hi,
       I am the user of wavedrom. Its a good tool for digital waveform edit. 
Due to my worse coding, the tool fetch huge of system memory and crash later.
Even I close the APP or Web Editor and open it again, it still wants to executes the pasted codes.
Then...crash again.
The problem is still existed even delete the APP and download APP from Web again.
I think that some information is stored in OS.
I can't use the tool anymore!!!! Unless I use another computer...
Is any way to solve this problem excluded OS reinstall..

My OS is Win7-64 bit

Thank you!!!

drom, 2017-03-27:
@Square2017 yes, you are right. This pain. Let me think of workaround, and long term solution after.

Square2017, 2017-03-28:
@drom Thank you

drom, 2017-03-28:
@Square2017 try deleting `C:\Users\username\AppData\Local\WaveDromEditor` folder.

Square2017, 2017-03-28:
@drom Yes, It works!!!!
Thank you!!!!!

closed 2017-03-30 by drom