wavedrom/wavedrom #173: Integrate wavedrom file to HTML

skazarynau, 2017-03-31:
I'm trying to integrate waveform from file to HTML page. This is the html file:
`<header>
  <script src="http://wavedrom.com/skins/default.js" type="text/javascript"></script>
  <script src="http://wavedrom.com/wavedrom.min.js" type="text/javascript"></script>
</header>
<body onload="WaveDrom.ProcessAll()">

<h2>WaveDrom Example</h2>

<p/>
<script type="WaveDrom" src="d:/dpkt_8_byte_1.json">
</script>
<p/>

</body>`
I see an error: Error: Unexpected token )

If I simply insert code from the file dpkt_8_byte_1.json, it works fine

drom, 2017-04-01:
`src="d:/dpkt_8_byte_1.json"` is illegal path in HTML.

closed 2017-04-01 by drom

skazarynau, 2017-04-01:
I also tried src="dpkt_8_byte_1.json". And I put this file to the same folder with html file. The result is the same.

drom, 2017-04-02:
@skazarynau that is correct. WaveDrom designed to work when WaveJSON description is inlined within `<script>` tag.

skazarynau, 2017-04-04:
There is the following workaround.
HTML file:
```
<html>
<head>
  <script src="http://wavedrom.com/skins/default.js" type="text/javascript"></script>
  <script src="http://wavedrom.com/wavedrom.min.js" type="text/javascript"></script>
  <script src="wave_file.js"></script>
</head>
</html>
```
JS file:
```
function readTextFile(file)
{
    var rawFile = new XMLHttpRequest();
    rawFile.open("GET", file, false);
    rawFile.onreadystatechange = function ()
    {
        if(rawFile.readyState === 4)
        {
            if(rawFile.status === 200 || rawFile.status == 0)
            {
                var allText = rawFile.responseText;
				document.write('\
					<body onload="WaveDrom.ProcessAll()">                                                      \
					<script type="WaveDrom" src="">                                                            \
					');
				document.write(allText);
				document.write('\
					</script>                                                                                  \
					</body>                                                                                    \
					');
            }
        }
    }
    rawFile.send(null);
}

readTextFile("dpkt_8_byte_1.json");
```
It works in Firefox, but not in Chrome

skazarynau, 2017-04-04:
To pass file name:
HTML:
```
<html>
<head>
  <script src="http://wavedrom.com/skins/default.js" type="text/javascript"></script>
  <script src="http://wavedrom.com/wavedrom.min.js" type="text/javascript"></script>
  <script type="text/javascript" id="fn" data-fn="dpkt_8_byte_1.json" src="wave_file.js"></script>
</head>
</html>
```
JS:
```
function readTextFile(file)
{
    var rawFile = new XMLHttpRequest();
    rawFile.open("GET", file, false);
    rawFile.onreadystatechange = function ()
    {
        if(rawFile.readyState === 4)
        {
            if(rawFile.status === 200 || rawFile.status == 0)
            {
                var allText = rawFile.responseText;
				document.write('\
					<body onload="WaveDrom.ProcessAll()">                                                      \
					<script type="WaveDrom" src="">                                                            \
					');
				document.write(allText);
				document.write('\
					</script>                                                                                  \
					</body>                                                                                    \
					');
            }
        }
    }
    rawFile.send(null);
}

var fn = document.getElementById("fn").getAttribute("data-fn"); 
readTextFile(fn);
```