wavedrom/wavedrom #174: Regarding adding source/destination feature in wavedrom

rapeta, 2017-05-31:
Hi!
  This is a great tool!  I am currently using it.
 
  However,  I am looking for source of the signal and destination where it is going to have better clarity.  Pasted the image below, what I am looking for. 

[image]

Please let me know if there is a example json file to comprehend that.
Best
-Rao

drom, 2017-06-01:
Hi Rao,
Just to understand your use case:
What vertical line represents?
Why SRC-A and DES-B shown two times?

rapeta, 2017-06-01:
Hi! Aliaksei

   Just indicating that both clk and data are coming from same source and going to same destination.  some signals might originate from different sources and similarly destinations could be different.

Best

-Rao