wavedrom/wavedrom #182: Script to import simvision csv?

electronicsguy, 2017-12-13:
Please describe your system environment before submitting your Issue.
Hi,
  great work! I was wondering if you have a script already to convert vcd or csv into the wavejson format? simvision exports this and it'd be great to be able to plot this data in wavedrom.

drom, 2017-12-13:
@electronicsguy no, I don't have such script.

electronicsguy, 2017-12-13:
The problem is that, the way vcd is, it puts a main timeline on top. Then every signal uses that timeline. Now if you have signal "a" changing at time "t", the clock signal "clk" may also have an entry at that time "t", but it'll be the same value carried over from time t-1. 

Now, this creates a dip when plotted in wavedrom, if the clk signal has adjacent values like "11"..

Plus it's a pain to recreate the timeline for every signal. I'm wondering if the code could be re-written, such that one signal like "clk" could be the base, and every other signal's transition will be relative to the times mentioned within the clk signal data. That would match the vcd format spec.

cmarqu, 2018-06-20:
For this, you kinda need the same process that you'd have to do to convert the event-based VCD into a cycle-based format for an Automated Test Equipment (ATE) - that process is usually called "cyclization", see e.g. https://www.edn.com/Home/PrintView?contentItemId=4380126 (more keywords related to this: STIL, WGL)
I have only found commercial tools for this so far.
It's probably easier to write a small testbench module that writes out WaveJson directly on each clock. Here is how coctb does it: https://github.com/potentialventures/cocotb/blob/master/cocotb/wavedrom.py

andresmanelli, 2019-07-13:
Hello, I am working in a python script "vcd2wavedrom". I am having good results for the moment:

[image]
=> From the example [here](https://en.wikipedia.org/wiki/Value_change_dump#Example_VCD_file)

[image]
=> Local file

Now the glitch comes from a `hscale: 0.1` (I deleted the `Math.round()`for this).
I needed this scale to stretch in cases where a signal changes state within the clock period (asynchronous reset for example). In this case, I have 10ns period, 1ns resolution.

Is there any way to include floating point scales? (If this is possible I'll open a specific issue)

Thanks for the great tool!

drom, 2019-07-13:
@andresmanelli WaveDrom tool and WaveJSON format were not designed to handle floating point scales. It was designed as a datasheet diagramming tool. I understand the need for simulation waveform viewer that would take VCD or another trace format.

I have started the waveform viewer project here: https://github.com/wavedrom/zoom
I think we also need proper streaming VCD parser: https://github.com/wavedrom/vcd/issues/1
Any contribution is welcome.

a-w-kelvin, 2024-09-16:
Hi, I know this is an old thread, but just wondering is there a utility/feature to dump out a VCD file from WaveJSON ?