wavedrom/wavedrom #183: Transitioned Data Cell with Continuation Squiggle

Remillard, 2018-01-09:
I've been experimenting with the web variation on this.  I assume it's kept up to date with the GitHub repository.

I don't know if this exists, but if not, maybe it could be a feature request.

Consider the following:
```
{
    "signal": [
        {"name": "ADDR LATCH", "wave": "10....|.....1.........|......"},
        {"name": "DATA LATCH", "wave": "1.....|..........0....|.....1"},
        {"name": "SCLK",       "wave": "l..nn.|...l........nn.|...l.."},
        {"name": "SDOUT",      "wave": "0..=======0........=======0..", "data": ["0", "1", "2", "...", "29", "30", "31", "0", "1", "2","...", "29", "30", "31"]},
        {"name": "SDIN",       "wave": "0..................=======0..", "data": ["0", "1", "2", "...", "29", "30", "31"]}
    ]
}
```
This renders: 
[image]

What I would like is to have a way to show that middle bit (the one labed "...") as a blank cell with an aligned "time-skip" squiggle just like above and below.  If I use a time-skip character `|` the preceding bit does not show transition like this does.  To demonstrate:

[image]

You'll note that the 2 bit looks like it continues which is not true.  I just need a way to show a number of unlabled bits in the same fashion in the middle.  Something akin to this drawn in Visio (my previous way of handling timing diagrams):

[image]

If there's a way to do this, please let me know.  If there isn't currently a way to do this, perhaps it could be a feature in the future.  Thanks!

jeras, 2018-02-01:
Hi @Remillard,

I encountered the same problem.
Do you have any ideas how to encode the behavior in json files? WaveDrom JSON format is very compact but not easily extensible. Some generalization is needed before this can be implemented.

The current behavior is to repeat the last state and to add the squiggle separator. The main purpose of  squiggle separators is to skip a static segment of the waveform. Repeating the last state is fine for static waves.

Cases where squiggle separators could be used are:
- static signal (already implemented as described above)
- periodic signal (period could be larger then one step)
- algorithmic sequence (increment, decrement, ...)
- random (can be handled with don't care values)

# Proposal:

There already is the dot `.` which is used to repeat the last state. First add a character that will draw the complement of the last state, I propose `~` which is used as a bitwise negator in Verilog. Second add a character that would complement the current value and draw the squiggle separator (I think it named "gap" in the original wiki documentation), I propose `!` which is used as logic negator in Verilog and is visually similar to the existing gap operator.

This would be a list of some states and their complements:
* `0` - `1`
* `=` - `=` (itself)
* `2`, `3`, `4`, `5` - `=` (neutral bus value)
* `x` - `x` (itself)
* `.` - `~` (kind off)
* `z` - `z` (itself)
* `u` - `d`
* `|` - `!`

All combinations would have to be checked to be sure new operators behave as we would like them too, but since WaveDrom is not a perfect tool, just covering most use cases might be good enough.

Remillard, 2018-02-01:
Sounds a bit excessive for a change.  Everything else thus far works to my satisfaction.  And indeed in the case I noted, for the data line during a time-skip interval, I simply put in the text '...' to indicate unspecified data during the time skip.  Since that was text, it correctly indicated the bit transition just prior.  I also played around with adding a blank data (with transition) then a time skip character, and then another blank data to sort of "pad" around the middle, but honestly that didn't produce a result I cared for.

All I would propose is a new block that has a data transition plus time-skip squiggle.  character for a time skip block that doesn't repeat the prior block.  Perhaps a capital `I` or something.  Everything else seems fine to me.  Your proposal sounds a little more encompassing so you might start a fresh issue on it.

jeras, 2018-02-01:
My proposal is actually as simple as yours, but I was thinking it up while writing, so it is a bit long.

The current behavior of `|` is to repeat the last wave state, the same as the dot `.` operator.
I proposed a `!` operator that would not repeat the last wave state. But the question remains what should be used instead of the last wave state, and I think it should be the complementary value of the last wave state. The purpose of the table is to list complementary values not only for the bus operator `=`, bot for other wave states too.

jmarcelold, 2022-12-15:
I would like to know if this issue has generated a new feature request. I believe only adding a code to print this specific draw
[image]
would be enough. 
I liked the idea of using `!` to print that.

Ganesh-AT, 2023-12-10:
The 'gaps' feature released in v3.1.0 earlier this year provides a flexible solution for this.

Here is a [rendering](https://wavedrom.com/editor.html?%7B%20%20%20%22signal%22%3A%20%5B%0A%20%20%20%20%20%20%20%20%7B%22name%22%3A%20%22ADDR%20LATCH%22%2C%20%22wave%22%3A%20%2210..........1................%22%7D%2C%0A%20%20%20%20%20%20%20%20%7B%22name%22%3A%20%22DATA%20LATCH%22%2C%20%22wave%22%3A%20%221................0..........1%22%7D%2C%0A%20%20%20%20%20%20%20%20%7B%22name%22%3A%20%22SCLK%22%2C%20%20%20%20%20%20%20%22wave%22%3A%20%22l..nn.....l........nn.....l..%22%7D%2C%0A%20%20%20%20%20%20%20%20%7B%22name%22%3A%20%22SDOUT%22%2C%20%20%20%20%20%20%22wave%22%3A%20%220..%3D%3D%3D%3D%3D%3D%3D0........%3D%3D%3D%3D%3D%3D%3D0..%22%2C%20%22data%22%3A%20%5B%220%22%2C%20%221%22%2C%20%222%22%2C%20%22...%22%2C%20%2229%22%2C%20%2230%22%2C%20%2231%22%2C%20%220%22%2C%20%221%22%2C%20%222%22%2C%22...%22%2C%20%2229%22%2C%20%2230%22%2C%20%2231%22%5D%7D%2C%0A%20%20%20%20%20%20%20%20%7B%22name%22%3A%20%22SDIN%22%2C%20%20%20%20%20%20%20%22wave%22%3A%20%220..................%3D%3D%3D%3D%3D%3D%3D0..%22%2C%20%22data%22%3A%20%5B%220%22%2C%20%221%22%2C%20%222%22%2C%20%22...%22%2C%20%2229%22%2C%20%2230%22%2C%20%2231%22%5D%7D%0A%20%20%20%20%5D%2C%0A%20head%3A%7Btick%3A0%7D%2C%0A%20gaps%3A%20%27.%20.%20.%20.%20.%20.%20s%20.%20.%20.%20.%20.%20.%20.%20.%20.%20.%20.%20.%20.%20.%20.%20s%27%0A%7D) of the diagram in the first post with the gap over the ... entry.

I think we can close this issue.