wavedrom/wavedrom #194: JupyterLab compatibility

xesscorp, 2018-06-18:
WaveDrom will run in a Jupyter notebook, but not in JupyterLab. For security, JupyterLab disables the execution of arbitrary Javascript and requires the creation of an *extension*. WaveDrom won't be usable as the Jupyter project moves forward unless such an extension is built.

drom, 2018-06-18:
I am not a big user of Jupiter notebook. Could you recommend the "extension builder's guide"?

xesscorp, 2018-06-19:
Thanks for the quick response! I'm not an expert on what an extension is or how much effort it would take. I'll do some research and let you know.

drom, 2018-07-19:
I am working to make WaveDrom available with [Observable](beta.observablehq.com)
It is similar to Jupiter but JS instead of Pyton. Is it interesting for you?
https://github.com/drom/wavedrom/issues/196