wavedrom/wavedrom #198: Generating bipolar signals

superlou, 2018-07-24:
We've been running WaveDrom happily on Windows 10 for generating digital signal timing diagrams. However, in a new project, we need to document bipolar waveforms. Is it possible to do this with WaveDrom? While multiple analog voltage levels would be great, just three levels would be something we could get away with.

drom, 2018-07-26:
Could you add a drawing of the bipolar waveform? Is it a trinary signal, or some sort of differential?

superlou, 2018-07-27:
I think the two biggest cases would be bipolar encodings like AMI in the [example](https://en.wikipedia.org/wiki/Bipolar_encoding) below:

[image]

The other use case is return-to-zero encodings like in this [example](https://en.wikipedia.org/wiki/Return-to-zero):

[image]

It would also be useful for diagramming states of outputs from H-bridges, e.g. voltage to motors. However, I think that too much feature creep could lead to trying to graph analog voltages, which is probably beyond the scope of WaveDrom.

drom, 2018-08-02:
Yes, this seems like useful feature, but lay aside of current tool.
It can be implemented as part of analog waveforming story, once we decide what is the best way to represent analog descriptions.
Related to: https://github.com/drom/wavedrom/issues/68

dgengland, 2019-07-16:
You can sort of get this by using the 'z' element:
For the AMI signal in your example:
{ name: "AMI", wave: '1z0z.101z.0z'}

drom, 2019-07-17:
or we can introduce a new character `-` fro 0.5 level ?

Ganesh-AT, 2023-12-23:
@superlou My apologies for resurrecting a very old thread.

We are working on creating an extensible framework for WaveDrom, and I believe that we should be able to address the 'bipolar signals' requirement now with reasonable effort. We have already got trigonometric waveforms and 'async' waveforms (similar to what is supported by [SchemDraw](https://schemdraw.readthedocs.io/en/stable/elements/timing.html)) working with the new framework, and would like to add bipolar signals support also.

Can you please let us know how you might want to specify the diagram ? Something on the lines of:

```
{ 
  name: 'AMI', wave: [ 'bipolar',
    { 'seq' : '01101', 'type': ('ami' | 'rtz'), ... ??? }
  ]
}
```

What other attributes do you envisage being necessary for the bipolar waveforms? Do you want the colors to be configurable? If so, how would you like to specify that?