wavedrom/wavedrom #199: Github markdown support

AntonKrug, 2018-07-26:
Is there a way, or a petition to create pressure on github to inegrate it into their markup? For example python notebooks are fully supported with github and so many editors support wavedrom already, but it would be super useful to have it in github.

drom, 2018-07-27:
I don't know how to do that. I think that GitLab integration may be easier because it is opensource.
Also, I am porting wavedrom to observable notebook now, that may solve some use cases. https://github.com/drom/wavedrom/issues/196

drom, 2019-02-28:
The other option is to run `wavedrom` server https://github.com/wavedrom/server
But that would require someone to sponsor it.