wavedrom/wavedrom #201: DDR Clock with Arrows

mrberman87, 2018-08-01:
I am wondering if there could be a new feature for adding up and down arrows on a clock to signify DDR clock edges.

drom, 2018-08-02:
Good idea. And easy to implement.
I guess we would need two variations:
p -> P -> ?
n -> N -> ?
The hardest part is to choose a letters.

mrberman87, 2018-08-06:
You could maybe use pipes?

P| for starting with a positive edge or N| for starting with a negative edge.  Then the period as usual.

ejeffrey, 2021-11-18:
For my purposes D for DDR would be sufficient and you could just use a phase shift to change the polarity.  Most DDR clocks I am used to run continuously and you don't have to worry about which edge is first.  There might be use cases I have not considered, so including a negative polarity DDR does make sense.  'd' would be easy to remember but not exactly fit.  Otherwise I say use 'V' for a negative edge first DDR clock.

igokugrok, 2022-12-13:
Did this ever get implemented?

smarshall54, 2025-04-09:
> Did this ever get implemented?

`{name: 'ddr_clk', 		wave: 'HL' .repeat(8)} `

scale down with period: 0.5 but for diagramming behavior in a DDR system you likely want the edges on the default ticks anyway and not on half-ticks.