wavedrom/wavedrom #205: Feature: Timing ranges/arrows on top of the diagram

darkstar, 2018-09-15:
It would be neat to have a feature that allowed diagrams similar to that one:
[image]
Note the "T_s", "T_c", "T_{s0}", etc. labels and arrows at the top

clobybob28, 2020-02-14:
Any chance of this being implemented? I also see #51 which sounds similar, but I can't see the full conversation. Looks like it's an older issue from Google Code.

crdavis12, 2020-02-14:
Any chance of this being implemented? I also see #51 which sounds similar, but I can't see the full conversation. Looks like it's an older issue from Google Code.

drom, 2020-02-14:
Are you looking for Arrows?
https://wavedrom.com/tutorial.html#arrows

crdavis12, 2020-02-14:
No, I'm thinking more like dimensions on a drawing. The T_s and other labels at the top of @darkstar's diagram, that show time (or some other unit) between two events in the diagram.

jason-s, 2020-09-11:
You can do that now, just put a blank row:

```
{signal:  
   [
     {name: 'timing', node:'..CD....E.'},
   {name:'S1',wave:'1..0.......',node:'...B..'},
    {name:'S2',wave:'1.0.....1..',node:'..A.....F.'},
  ],
 config: {hscale:1},
 head:{
   tick:null,
 },
 edge:['A-C','B-D','C<->D T1',
       'E-F', 'D<->E T2'
 ],
}
```

[image]

It does require using up extra timing nodes though.

Maybe it would help to be able to specify the intersection of a node and a signal, for example `A-timing@A` rather than `A-C` and having to create an artificial node C that is in the same horizontal position as A but in another waveform's row.

adrienkohlbecker, 2021-10-03:
This MR might be helpful https://github.com/wavedrom/wavedrom/pull/337