wavedrom/wavedrom #206: Add overline to signal name

whc2001, 2018-10-14:
Please add a way to mark a signal name with overline as it's quite useful to express low-enabled signal. It's better to use a particular character to individually mark each character that needs to be overlined than overline the entire signal name (like [this](https://github.com/drom/wavedrom/issues/100)). For example, in Altium Designer, writing `R/W\` for pin name will result in [image]. Thanks.

drom, 2018-10-14:
Overline is supported with `<o>text</o>` 
for other markup see https://github.com/drom/tspan

whc2001, 2018-10-14:
Awesome, thanks!

closed 2018-10-14 by whc2001