wavedrom/wavedrom #207: Positive edge clock wave missing rising edge of clock when following high

engicoder, 2018-11-03:
The rising edge of a positive clock wave is replaced by the dashed line when following an 'h'. Example below:
`{signal: [
  {name: 'clk',  wave: 'hp...'},
],
}
`
[image]

engicoder, 2018-11-04:
After reviewing other bugs in more detail, I think this is the same issue as #162 
https://github.com/drom/wavedrom/issues/162

Ganesh-AT, 2019-10-13:
The work-around for #162 is fairly simple:

```
{name: 'clk', wave: 'hpp...'},
```

should give the desired waveform without rendering issues, as shown below.

That said, @drom should clarify on the intentions of where p,P,n,N can be used. As per WaveJSON documentation, it appears that they can only be at the beginning of a wave string. But, I can see it being useful for cases like what you are describing. The issue comes in the form of inconsistency (if your code is valid WaveJSON) - while your sample has a 'hp' being rendered with a proper transition between h and p, hP is not 

```
{signal: [
  {name: "clk_bug", wave: "hp...."},
  {name: 'clk', wave: 'hpp...'},
  {name: "clkP_bug", wave: "hP...."},
  {name: 'clkP', wave: 'hPP...'},
]}
```

renders the image below

[image]

I think hP should also have a transition similar to hp.

Supporting p,P,n, and N in sub-cycles is also not working unless we make modifications in gen-wave-brick.js - a file that hasn't been touched in 4 years.