wavedrom/wavedrom #215: I'm dumb and I can't find the manual.

ScumCoder, 2019-02-16:
Hello and thanks for the tool.
I tried looking both on the [official website](https://wavedrom.com)  and here on Github, but I can't find the WaveJSON documentation - only the [Hitchhiker's](https://wavedrom.com/tutorial.html) [Guides](https://wavedrom.com/tutorial2.html). While they are good to get a grasp on the syntax, I would much prefer a comprehensive and structured document. As a bare minimum, I would like to see a full list of all possible letters that can be specified in the `wave` node (I had to gather them from examples).

drom, 2019-02-18:
Here are some relevant documents:
https://github.com/wavedrom/schema/blob/master/WaveJSON.md
https://github.com/wavedrom/schema/blob/master/wavedrom.json

ScumCoder, 2019-02-18:
Exactly what I was looking for.
Maybe it makes sense to drop this link at the end of Hitchhiker's Guide?

closed 2019-02-18 by ScumCoder

ScumCoder, 2019-02-18:
I spoke too soon :)
Even this MD document does not list ALL possible waveforms. There are also `h`, `l`, `H` and `L`.