wavedrom/wavedrom #217: Use of getBBox()

sebunger, 2019-03-10:
Hi,

I'm trying to print an HTML file containing Wavedrom diagrams using [Prince](https://www.princexml.com/). However, this fails because Prince does not support the getBBox() method. Below is the response I got from them:

> It appears to be calling the getBBox() method, which Prince does not 
> support. We can stub that out like this:
> 
> Element.prototype.getBBox = function() { return {width: 100}};
> 
> Then the script runs, but it probably will not produce useful results if 
> it is expecting to be able to get the final element positions and sizes, 
> as Prince runs all JavaScript before doing layout.
> 
> Until we support full reflow capability, which might not happen until 
> next year, it might require a bit of work to support this script, unless 
> there are some specific width values that can be inserted to make it get 
> the correct output, I'm not sure.

I notice that getBBox() is only called in three places and always to get the width. My question is whether there is any other way to get or approximate the width that could be used in case getBBox() is not supported. Even a hack not fit to include in the standard code that produces something that isn't quite right would be a step forward as I could easily select the hacked JS for printing and the standard JS for showing in a browser.

I'd appreciate any ideas.

Cheers,
Seb

drom, 2019-03-12:
Hi @sebunger 
Yes, getBBox used only to measure text box width, and there are different ways to get an approximate number. This particular issue is something that I wanted to fix for a very long time. Removing getBBox will help wavedrom engine running outside of the browser.

drom, 2019-05-17:
Should be fixed in wavedrom v2.1.0. @sebunger, please try

sebunger, 2019-05-19:
Just tried and it looks great! I think this issue is resolved.

Thanks

closed 2019-05-19 by sebunger