wavedrom/wavedrom #233: Signal shifted by phase is not properly drawn

AlessioTonioni, 2019-05-24:
I'm using the web editor.

The following json
```
{signal: [
  {name: 'clk', wave: 'P..'},
  {name: 'EN', wave: 'X01', phase: 0.5},
]}
```
produce this visualization:
[image]

However, if I change the phase to something else:
```
{signal: [
  {name: 'clk', wave: 'P..'},
  {name: 'EN', wave: 'X01', phase: 0.2},
]}
```
The EN signal does not properly start from the beginning of the plot
[image]

drom, 2019-05-31:
`X` is not supported. Use `x` instead.

AlessioTonioni, 2019-06-01:
> 
> 
> `X` is not supported. Use `x` instead.

Besides 'X' and 'x', with any phase that is not a multiple of 0.5 the signal does not start from the beginning of the plot but it is slightly shifted to the right.
If you compare the two images I have uploaded above you should see how with phase 0.5 the signal start from t=0 while with a different phase it starts from an arbitrary position.

Another example: 
```
{signal: [
  {name: 'OK', wave: '010....', phase: 0.5},
  {name: 'NOT_OK', wave: '010....', phase: 0.1}
  ]
}
```
[image]

drom, 2019-06-01:
wavedrom diagram is constructed from standard building blocks (bricks) of fixed size.
With period=1 there are 2 bricks per period.
With period=2 there are 4 bricks.
Choosing `narrow` skin and period > 1 will give you more phase accuracy.

Like this: 

```js
{
  signal: [
    {name: 'OK0', wave: '010....', period: 2, phase: 1.5},
    {name: 'OK1', wave: '010....', period: 2, phase: 1},
    {name: 'OK2', wave: '010....', period: 2, phase: 0.5},
    {name: 'OK2', wave: '010....', period: 2},
  ],
  config: {skin: 'narrow'}
}
```