wavedrom/wavedrom #234: Gated clock implementation - glitch on ungate

ghanashyamprabhu, 2019-05-31:
Am using the below code to implement a gated clock waveform - the ungated clock, the gated clock and the clock enable are shown. Why do I see a glitch at the clock cycle where clock is gated again second time?

```
{ "signal" : [
   { "name": "clk",        "wave": "p.......|........" },
  { "name": "clk_enable",  "wave": "0..1....|...0...." },
  
   { "name": "gated_clk",  "wave": "0...p...|....0..." }
]}
```
Can you please help resolving this issue? 

[image]

Regards
ghanashyam

drom, 2019-05-31:
Just use `l` instead of `0`

ghanashyamprabhu, 2019-06-01:
Below is the updated code with `1` instead of `0` - The clock pulls high which is not what I intend on clock gate. On clock gate, the clock needs to park low. 
```
{ "signal" : [
   { "name": "clk",        "wave": "p.......|........" },
  { "name": "clk_enable",  "wave": "0..1....|...0...." },
 
   { "name": "gated_clk",  "wave": "0...p...|....1..." } // changed 0 to 1 
]}

```
[image]

drom, 2019-06-01:
Not `1` as ONE but `l` as (Low case LETTER L)

ghanashyamprabhu, 2019-06-02:
Thanks that resolves, Pasting the code here for completion
```
{ "signal" : [
   { "name": "clk",        "wave": "p.......|........" },
  { "name": "clk_enable",  "wave": "0..1....|...0...." },
 
   { "name": "gated_clk",  "wave": "0...p...|....l..." }
]}
```
[image]

closed 2019-06-02 by drom