wavedrom/wavedrom #235: Allow modification of previously generated PNG or SVG images

corna, 2019-06-07:
It would be great to allow the edit of previously generated waveforms.Currently, this is possible only by keeping a copy of the JSON source.

Is it possible to store the source JSON in the image metadata, so that it can be automatically extracted by wavedrom? With PNGs it should be possible by using [textual data](https://www.w3.org/TR/PNG/#11textinfo) (even unicode with iTXt) and SVG should support it within the [\<metadata\> tag](https://www.w3.org/TR/SVG11/metadata.html).

Thanks

drom, 2019-06-09:
Awesome idea @corna. In SVG, do you see it as one of the standard "metadata" types? RDF, OWL, SKOS,... or just blob of multiline text? Could you give an example of how it should look like?

corna, 2019-06-23:
Unfortunately I have no experience on SVGs other than just using them.

maxslug, 2023-04-12:
I was poking around on how to do ths with PNG.  Here are some breadcrumbs
- https://stackoverflow.com/questions/48243115/how-do-i-insert-a-png-comment-block-when-saving-an-html5-canvas-using-javascript
- https://stackoverflow.com/questions/44489503/how-can-i-append-a-text-chunk-into-a-canvas-data-array-and-save-it-as-a-png-file
- https://github.com/xgds/pngtoy