wavedrom/wavedrom #240: How to do that？

PYGC, 2019-07-03:
Please describe your system environment before submitting your Issue.

[image]

drom, 2019-07-08:
```js
{signal: [
  {name: 'CLK', wave: 'p..|.|....'},
  {name: 'RXD', wave: 'x=.|=<=|>===x', data: ['PREAMBLE', 'SFD', '', '', '', 'FCS']},
  {name: '???', wave: '0..|.|....'}
]}
```
Here is the trick:

[image]

drom, 2019-07-08:
I was thinking of adding `#` symbol to this, particularly popular case. :thinking: 

Related to #213

closed 2020-03-23 by PYGC