wavedrom/wavedrom #242: Clock frequency change

ghanashyamprabhu, 2019-07-12:
Is there a way I can represent a clock frequency switch in a clock signal without glitchy behavior 

`{ "name": "clk",  "wave": "P.........npnpnpnpnpnpnpnp"}, `
The above code has a clock that changes its frequency from F to F/2 and the change happens at the negedge of the clock signal. This code has some glitchy signal rather than a clean clock as shown in the snapshot below

[image]

-ghanashyam

drom, 2019-07-13:
Use `l`, `L`, 'h`, `H` chars:

```js
{signal:[{name:'clk',wave:'P.....lHlHlHlH'}]}
```
[image]

ghanashyamprabhu, 2019-07-13:
Note the change of period at the negedge of the clk. With the  code that
you have suggested, at the point of frequency change, the negative phase of
the clock has the (old period)/2 as well as (new clock period)/2

drom, 2019-07-13:
```
{signal:[{name:'clk',wave:'P.....lHlHlHlH'}]}
```
[image]

Closer?

ghanashyamprabhu, 2019-07-14:
```
{signal:[{name:'clk',wave:'P.....<l>HlHlHlH'}]}
```

does the trick! Thanks much, this is useful. I couldn't find this on the tutorial page though..

drom, 2019-07-14:
Yes, I have created a separate issue to document sub-cycle feature here: https://github.com/wavedrom/wavedrom/issues/243

closed 2019-07-14 by drom

ghanashyamprabhu, 2019-07-14:
While we are at it, can I also make a separate signal which is a pulse occurring from the posedge to posedge at the clock cycle where the frequency changed in the w/f that I posted in question?

drom, 2019-07-14:
Something like this?

[image]HlHlH'},{name:'change',wave:'0..h<.>l....'}]}"/>

In theory, it should look like this:

```js
{signal:[
  {name:'clk',    wave:'P...<l>HlHlH'},
  {name:'change', wave:'0..h<.>l....'}
]}
``` 

But in practice, it is broken.

lilyoba, 2024-01-23:
> ```
> {signal:[{name:'clk',wave:'P.....lHlHlHlH'}]}
> ```
> 
> [image]
> 
> or
> 
> ```
> {signal:[{name:'clk',wave:'P.....<l>HlHlHlH'}]}
> ```
> 
> [image]
> 
> Closer?

I am wondering how to generate this kind of waveform, thank you very much