wavedrom/wavedrom #246: Prince warning caused by WaveDrom generated SVG

jmilstead-sifive, 2019-08-15:
An SVG generated by WaveDrom produces the following warning in Prince:
 
`warning: unsupported properties: enable-background
`

The workaround was to open the SVG in a text editor and delete all 5 instances of  `enable-background:accumulate`

nschwass, 2020-05-20:
I'm going to resuscitate this issues page with some images to corroborate Jeff's gripes here. The figure below shows what happens in .pdf output when you remove those background-accumulate. Although this seems to be a prince problem vs. a wavedrom problem, it would be nice to have a workaround for the time being.
[image]

drom, 2020-05-20:
I will fix it.

commit d63bc55c (2020-05-20): fixes #246; delete enable-background:accumulate
closed 2020-05-20 by drom